weirdgyn/viastitcher #1: Not compatible with KiCAD 6.0

bookshiyi, 2021-11-23:
Will it be updated in the future?

[image]

**Information：**

Application: KiCad PCB 编辑器 (64-bit)
Version: (6.0.0-rc1-45-gbfad40bdcc), release build
Libraries:
    wxWidgets 3.1.5
    libcurl/7.78.0-DEV Schannel zlib/1.2.11
Platform: Windows 10 (build 22000), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Date: Nov 17 2021 12:01:05
    wxWidgets: 3.1.5 (wchar_t,wx containers)
    Boost: 1.76.0
    OCC: 7.5.0
    Curl: 7.78.0-DEV
    ngspice: 35
    Compiler: Visual C++ 1929 without C++ ABI
Build settings:
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON

weirdgyn, 2021-12-05:
thank you for the report

weirdgyn, 2022-01-07:
> I came here to check before making the big jump. This is one very
> important function missing from 6.0 :) Thank you for this plugin

I still have to install the new 6.0 (just downloaded it at this time).
I'll check it and hopefully fix it in the next few days.t

weirdgyn, 2022-01-07:
> thank you for the report

of course yes I'll fix it as long as I get 6.0 running on my notebook ... thanks for your patience...

weirdgyn, 2022-01-07:
I did some fixes to work with KiCAD 6.0. Did not make extensive testing but it looks like it's working now. Please let me know.

closed 2022-01-07 by weirdgyn

tchiwam, 2022-01-07:
Using it now, so far so good, thanks a lot!