weirdgyn/viastitcher #2: Not working in Kicad 6.0 on linux

dimonDDL, 2022-01-27:
I tried using the plugin after upgrading but after clicking "ok" nothing happens. I launched kicad using console to see what the output is and got the following:
```Traceback (most recent call last):
  File "/home/dmytro/.local/share/kicad/6.0/scripting/plugins/viastitching-master/viastitching_dialog.py", line 307, in onProcessAction
    self.FillupArea()
  File "/home/dmytro/.local/share/kicad/6.0/scripting/plugins/viastitching-master/viastitching_dialog.py", line 278, in FillupArea
    if self.area.HitTestInsideZone(p):
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 14913, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, ZONE, name)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 80, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'ZONE' object has no attribute 'HitTestInsideZone'
```
Any idea what could be causing this and how to fix it?

KarlosA19, 2022-05-17:
Same problem here. 
I'm using kicad 6.0 and Ubuntu 20.04.4 LTS.

danwood76, 2022-08-08:
A couple of changes are required to make it work in KiCAD 6, attached is a patch.

[viastitcher-kicad-6.zip](https://github.com/weirdgyn/viastitching/files/9282977/viastitcher-kicad-6.zip)

weirdgyn, 2022-08-09:
What about a pull request?

danwood76, 2022-08-09:
I have uploaded a pull request.

The VIA stitching wasn't perfect, but much easier to clean up than filling the whole board with the "create array method". I had never used this plugin before yesterday so no idea if it worked better in previous versions.

closed 2022-08-09 by weirdgyn