weirdgyn/viastitcher #6: delete failed

HaydenHu, 2022-09-03:
"Clear only plugin placed vias" cannot be deleted after being selected, and can be deleted when deselected
[image]

win10
kicad6.0.7

weirdgyn, 2022-09-04:
Yes I know ... the problem arise from an 'update' on the latest KiCAD Python API which modified the board/vias structure and now it is impossibile to 'sign' board objects ... it's pratically impossible at the moment to track vias . I'm still keeping the function in the eventuality that further API updates give another way to sign objects.

closed 2022-09-04 by weirdgyn

HaydenHu, 2022-09-04:
> 是的，我知道...问题是由最新的 KiCAD Python API_更新_引起的，该 API 修改了电路板/通孔结构，现在不可能对电路板对象进行_签名...目前几乎不可能__追溯_创建的通孔。我仍然保留该功能，以防进一步的 API 更新提供另一种签名对象的方式。

https://github.com/jsreynaud/kicad-action-scripts
You can refer to this plugin to put the added vias in a group for manual removal

weirdgyn, 2022-09-05:
Nice hint. I think it works fine now.... take a look.