weirdgyn/viastitcher #7: Net conflicts

HaydenHu, 2022-09-05:
It cannot automatically evade other nets
[image]

weirdgyn, 2022-09-06:
that's strange... 
[image]

HaydenHu, 2022-09-06:
> 这很奇怪......这就是它应该如何工作（至少在我的设置中）：
> 
> [image]
> 
> 也许您将_间隙_参数保留为**0**？
> 
> [image]
> 
> 此值将禁用间隙检查），如[“工作原理](https://github.com/weirdgyn/viastitching#how-it-works)”部分所述。

Clearance settings to 0 and others have tried

[test.zip](https://github.com/weirdgyn/viastitching/files/9495219/test.zip)

weirdgyn, 2022-09-06:
Have you tried setting clearance to a NON 0 value ? Which values you used?
Via size/drill values? Spacing?
Anyway at the moment the overlapping check still need to be refined (as stated in docs).

weirdgyn, 2026-08-21:
Closing as resolved/superseded by the later CheckOverlap overhaul, particularly PR #44, together with the pad and overlapping-zone fixes in PRs #41 and #37. If net conflicts remain reproducible on current master, please open a new issue with the board file, KiCad version, and clearance settings.

closed 2026-08-21 by weirdgyn