weirdgyn/viastitcher #14: no via implanted

adanazem, 2022-11-21:
For some reason, I do not have the fill option to check and regardless of the settings, no vias are implanted.
[image]
[image]

weirdgyn, 2022-11-22:
could you share your pcb file or a test pcb file where to reproduce this
issue?

adanazem, 2022-11-22:
here's a link to a pcb test file
https://drive.google.com/file/d/1_vREOaaWlCKVUErPSCs9YOCw9udgh_Uw/view?usp=sharing

I tried with multiple pcb files and always receive the same error.

artemkotenko96, 2022-11-30:
I had the same issue and resolved it by using only one filled zone for Top and Bottom Layers. Just select both F.Cu and B.Cu in Copper Zone Properties.

adanazem, 2022-11-30:
That worked. Thank you!

closed 2022-11-30 by adanazem