weirdgyn/viastitcher #15: Execution time

immortal-sniper1, 2022-11-22:
So is is normal for the script to take hours to run on a 4 layer board?
I did run it with great effect on 2 layer boards and it was fine. but not it takes ages, i can see that some vias were placed both in layout and 3d model but kicad is frozen. 
Also on a side note is there a way to tell the script not to put vias in certain areas so that i can later remove that zone and stitch with another net?

weirdgyn, 2022-11-22:
Il giorno mar 22 nov 2022 alle ore 14:15 immortal_sniper1 <
***@***.***> ha scritto:

> So is is normal for the script to take hours to run on a 4 layer board?
>
Nope the script is usually quite fast and I don't think there's a way to
make it faster since the placement is made
scanning the area horizontally and vertically and checking the room for the
via...

> I did run it with great effect on 2 layer boards and it was fine. but not
> it takes ages, i can see that some vias were placed both in layout and 3d
> model but kicad is frozen.
>
Could you share a pcb file where I can reply the same issue?

> Also on a side note is there a way to tell the script not to put vias in
> certain areas so that i can later remove that zone and stitch with another
> net?
>
This depends on what you mean with area.... if it is a named object that is
accessible through KiCAD python API we can find a way to access it and
check if via falls inside its borders and so on...

Again if you can provide a PCB file with such an element I can check if
this feature can be *implemented*.

weirdgyn, 2022-11-23:
>
> https://github.com/immortal-sniper1/Bird_Detector/tree/master/Bird_detector_V1

I'll make a try on this board and I'll let you know..
>
> I also tried on another board with a net that is not gnd and it did not
> trigger
again without a 'problematic' pcb file the reply of issue is uncertain...

> regarding the area i want to not put vias inside a SMA connector or small
> areas like that
We have to find a way to 'classify' such kind of components. Not too
difficult I'll think about it.

immortal-sniper1, 2022-11-23:
again without a 'problematic' pcb file the reply of issue is uncertain.....

ill upload that board soon too it is not MIT that is why i have to put some other license on it and it is a bit more complicated but i think ill be able to get it done  but realistically i am not sure at what time 

regarding classification maybe this: if it is a THT and pitch is 2-3 times via spacing dont place via there ( just what i can up on the fly but i think this more or less can work)

weirdgyn, 2022-11-29:
Still working on it.

immortal-sniper1, 2022-12-14:
tried it again on a few boards and again it went wild with RAM and time 

https://github.com/immortal-sniper1/cross_band_handy_walkie_talkie/tree/master/HW

weirdgyn, 2026-08-25:
This issue refers to a performance / memory problem observed with an older
version of ViaStitching on complex and multilayer boards.

The placement and overlap-checking code has changed substantially since the
original report, including several fixes to zone, pad and track collision
handling which removed unnecessarily broad geometric checks.

Current testing with recent KiCad/ViaStitching versions, including multilayer
boards and hundreds of generated vias, no longer shows the extreme execution
times or RAM usage described here.

Since the original problem dates back to 2022 and has not been reported again
against the current implementation, I'm closing this as completed.

If excessive execution time or memory usage can still be reproduced with the
current version, please open a new issue including:
- exact KiCad version;
- ViaStitching version/commit;
- PCB file or minimal reproducible example;
- approximate via count / spacing;
- execution time and observed memory usage.

closed 2026-08-25 by weirdgyn