weirdgyn/viastitcher #17: Geting stuck

immortal-sniper1, 2023-01-14:
So after running the scrips of a few boards i found this pattern:
It works well only on simple 2 layer boards, if it has more then 2 layers or is fairly complex it gets stuck and computes for hours.

BUT also sometimes if the runtime is long enough it finds a pattern and  layers it out visually and if you wit enough it will be auto saved or something and after you kill the task and reload the project it will still be there. 

Also would using random help in these cases? is it a lighter script in terms of resources?

immortal-sniper1, 2023-01-14:
Update so acourding the the script log it finished in like 2-3min BUT it kept running for another 2hours for some reason...

PS the log is 44x MB large if i save it in notepad and it mostly made out of this line:
12:52:03: exception: missing HitTestInsideZone: To Be Fixed
12:52:03: exception: missing HitTestInsideZone: To Be Fixed
12:52:03: exception: missing HitTestInsideZone: To Be Fixed
12:52:03: exception: missing HitTestInsideZone: To Be Fixed
12:52:03: exception: missing HitTestInsideZone: To Be Fixed
12:52:03: exception: missing HitTestInsideZone: To Be Fixed
12:52:03: exception: missing HitTestInsideZone: To Be Fixed

weirdgyn, 2026-08-21:
Closing as obsolete and covered by #15. The repeated missing HitTestInsideZone path reported here is no longer present in the current implementation. Please continue current performance reports in #15 with a reproducible board and KiCad version.

closed 2026-08-21 by weirdgyn