weirdgyn/viastitcher #24: Doesn't do anything: NotImplementedError: Wrong number or type of arguments for overloaded function 'new_VECTOR2I'.

readmodifywrite, 2023-05-31:
Pressing the OK button results in nothing.  No messages, no vias, nothing.

If I open the Python console and then hit OK, I'll get some errors in another window:

Traceback (most recent call last):
  File "/home/jeremy/.local/share/kicad/6.0/scripting/plugins/viastitching-master/viastitching_dialog.py", line 456, in onProcessAction
    self.FillupArea()
  File "/home/jeremy/.local/share/kicad/6.0/scripting/plugins/viastitching-master/viastitching_dialog.py", line 363, in FillupArea
    p = pcbnew.VECTOR2I(xp, yp)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 5286, in __init__
    this = _pcbnew.new_VECTOR2I(*args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'new_VECTOR2I'.
  Possible C/C++ prototypes are:
    VECTOR2< int >::VECTOR2()
    VECTOR2< int >::VECTOR2(wxPoint const &)
    VECTOR2< int >::VECTOR2(wxSize const &)
    VECTOR2< int >::VECTOR2(int,int)
    VECTOR2< int >::VECTOR2(VECTOR2< int > const &)

Traceback (most recent call last):
  File "/home/jeremy/.local/share/kicad/6.0/scripting/plugins/viastitching-master/viastitching_dialog.py", line 456, in onProcessAction
    self.FillupArea()
  File "/home/jeremy/.local/share/kicad/6.0/scripting/plugins/viastitching-master/viastitching_dialog.py", line 363, in FillupArea
    p = pcbnew.VECTOR2I(xp, yp)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 5286, in __init__
    this = _pcbnew.new_VECTOR2I(*args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'new_VECTOR2I'.
  Possible C/C++ prototypes are:
    VECTOR2< int >::VECTOR2()
    VECTOR2< int >::VECTOR2(wxPoint const &)
    VECTOR2< int >::VECTOR2(wxSize const &)
    VECTOR2< int >::VECTOR2(int,int)
    VECTOR2< int >::VECTOR2(VECTOR2< int > const &)

Traceback (most recent call last):
  File "/home/jeremy/.local/share/kicad/6.0/scripting/plugins/viastitching-master/viastitching_dialog.py", line 456, in onProcessAction
    self.FillupArea()
  File "/home/jeremy/.local/share/kicad/6.0/scripting/plugins/viastitching-master/viastitching_dialog.py", line 363, in FillupArea
    p = pcbnew.VECTOR2I(xp, yp)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 5286, in __init__
    this = _pcbnew.new_VECTOR2I(*args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'new_VECTOR2I'.
  Possible C/C++ prototypes are:
    VECTOR2< int >::VECTOR2()
    VECTOR2< int >::VECTOR2(wxPoint const &)
    VECTOR2< int >::VECTOR2(wxSize const &)
    VECTOR2< int >::VECTOR2(int,int)
    VECTOR2< int >::VECTOR2(VECTOR2< int > const &)

splitice, 2023-07-02:
Ive got a similar issue:
```
Traceback (most recent call last):
  File "C:\Users/splitice/Documents/KiCad/6.0/scripting\plugins\viastitching\viastitching_dialog.py", line 456, in onProcessAction
    self.FillupArea()
  File "C:\Users/splitice/Documents/KiCad/6.0/scripting\plugins\viastitching\viastitching_dialog.py", line 368, in FillupArea
    if self.area.HitTestFilledArea(layer, p, 0):
  File "C:\Program Files\KiCad\6.0\bin\Lib\site-packages\pcbnew.py", line 12507, in HitTestFilledArea
    return _pcbnew.ZONE_HitTestFilledArea(self, aLayer, aRefPos, aAccuracy)
TypeError: Wrong number or type of arguments for overloaded function 'ZONE_HitTestFilledArea'.
  Possible C/C++ prototypes are:
    ZONE::HitTestFilledArea(PCB_LAYER_ID,wxPoint const &,int) const
    ZONE::HitTestFilledArea(PCB_LAYER_ID,wxPoint const &) const
```

mpieper, 2023-07-22:
Okay several things.

1.  It appears KiCad 6.0 does not have a vector2I input to HitTestFilledArea (I could only find the documentation online, and the docs say it should work, but the code's error warnings speak for themselves).  Forcing it to a wxPoint gets past that (are there ways to check for what instance of Kicad you're running? e.g. version<7.0 -> use wxPoint)
2.  Your code checks for intersections with ZONES and FP_ZONES, whereas for my use case that's exactly what I want to use it for (e.g. filling a copper zone).  Commenting that out fixed it, but I don't know why you were checking for intersections there in the first place. (def CheckOverlap(self, via): ->about line 300).  
3.  There's a bug (?) in randomize, where your divisions might not be integers (which are required for wxPoint instantiation).  Casting after the calc fixes that. (if self.randomize: -> about line 360)
4. Nice work and thank you for writing this in the first place.

matthuszagh, 2023-07-30:
Seeing the same issue on kicad 7.0.2 when I use "randomize". If I deselect randomize, no vias are placed.

ehammer33, 2023-09-25:
Just reiterating what @mpieper figured out:

replace 259 with:  the_distance = norm(pcbnew.VECTOR2I(p2) - pcbnew.VECTOR2I(p1)) 
replace 356 with:  xp = x + random.uniform(-1, 1) * int(step_x / 5)
replace 357 with:  yp = y + random.uniform(-1, 1) * int(step_y / 5)

comment-out lines 317-319.

comment-out lines 362-364.
Then just have:
if(hasattr(pcbnew, 'wxPoint')):
    p = pcbnew.wxPoint(xp, yp)
    
replace line 368 with:  if self.area.HitTestFilledArea(layer, pcbnew.VECTOR2I(p), 0):
replace line 370 with:  via.SetPosition(pcbnew.VECTOR2I(p))

weirdgyn, 2026-08-21:
Closing as resolved/superseded on current master. Coordinates passed to VECTOR2I are now explicitly converted to integers, including randomized coordinates. If a current KiCad version still produces a VECTOR2I or HitTestFilledArea traceback, please open a new issue with the exact version and full traceback.

closed 2026-08-21 by weirdgyn