weirdgyn/viastitcher #26: Via Stitching places 0 vias

n8ur, 2023-11-13:
I have a 4-layer board with 3 layers having ground pour and one layer having Vcc pour.  I am trying to stitch ground vias and no matter what settings I try ViaStitching returns successfully but reports placing 0 vias.  I have other boards where stitching works properly, so it must be something about the board settings for this board.  I've tried various clearances and hold/via sizes but nothing changes the result.

Any suggestion what I should be looking for?

Running KiCad Application: KiCad PCB Editor x86_64 on x86_64

Version: 7.0.9-7.0.9~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 6.0.0
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.16

Platform: Linux Mint 21.2, 64 bit, Little endian, wxGTK, cinnamon, x11

Build Info:
	Date: Nov  8 2023 19:01:25
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.88.1
	ngspice: 38
	Compiler: GCC 11.4.0 with C++ ABI 1016

Build settings:
	KICAD_SPICE=ON

KarolBorecki, 2024-10-11:
same issue here
[image]

jmole, 2024-12-25:
I've updated the plugin to KiCAD 8 here: https://github.com/jmole/viastitching/tree/master which includes some fixes to multi-layer polygons.

give it a try if you have some time and let me know if you still have the issue

weirdgyn, 2026-08-25:
Retested with the current version using a representative 4-layer setup:

- one multilayer GND zone on F.Cu + In1.Cu + B.Cu
- one VCC zone on In2.Cu
- overlapping XY extents

ViaStitching now generates the via pattern correctly; the original "0 vias" behavior is no longer reproducible.

A relevant fix was introduced in commit
cfda9f9daccac9980f9788e4036638758c2907c3, which corrected two cases that could cause "No Vias Implanted":

- same-net zones being treated as obstacles;
- overlapping zones being checked using bounding-box intersection instead of actual filled copper.

Subsequent multilayer / filled-copper handling has also been improved.

Closing as completed.

If anyone can still reproduce a zero-via result on a current version, please provide the exact KiCad version, ViaStitching version and a minimal `.kicad_pcb` test case.

closed 2026-08-25 by weirdgyn