weirdgyn/viastitcher #29: Clearance

MaiTheSan, 2024-02-19:
when set value in a Clearance box, it do nothing and it can't generate via.
[image]

ODYYYYYY, 2024-04-23:
I have a same problem here.

weirdgyn, 2026-08-24:
Can you check it against the latest release and tell me if it's fixed now?

weirdgyn, 2026-08-25:
Fixed in commit [c1c6ecb7f0d2d17c15000aad67b3bb95b23694a4](https://github.com/weirdgyn/viastitching/commit/c1c6ecb7f0d2d17c15000aad67b3bb95b23694a4).

The issue was caused by subtracting SWIG geometry objects of different types
(`wxPoint` and `VECTOR2I`) inside `CheckClearance()`.

The distance calculation now operates directly on the numeric `x`/`y`
coordinates using `math.hypot()`, avoiding any dependency on the concrete
geometry wrapper type.

The original issue was reported on KiCad 7. I cannot currently retest that
exact version, but the failure mode is directly addressed by the fix.

Regression testing on KiCad 10 was also performed successfully:
- default settings: 485 vias placed
- 1 mm clearance: 284 vias placed
- 2 mm clearance: 64 vias placed
- more restrictive test settings: 3 vias placed, all in valid free areas

No regressions in via placement or clearance handling were observed.

Closing as completed. Further testing on KiCad 7/8 is welcome; if the problem
can still be reproduced, please reopen the issue with the exact KiCad version
and test case.

weirdgyn, 2026-08-25:
Retested with the current code and I can no longer reproduce this issue.

Clearance handling was explicitly tested on KiCad 10 using the same board and via settings while varying only the clearance:

default settings / no additional clearance: 485 vias
1 mm clearance: 284 vias
2 mm clearance: 64 vias

Via placement therefore responds correctly to the configured clearance and progressively excludes positions close to zone boundaries and other obstacles.

Some clearance-related compatibility code has also been updated since the original report, including the recent wxPoint / VECTOR2I fix in CheckClearance() (c1c6ecb7f0d2d17c15000aad67b3bb95b23694a4).

Since the original problem is no longer reproducible with the current version, I'm closing this as completed.

If anyone can still reproduce a case where setting a non-zero clearance results in zero vias, please open a new issue (or request reopening this one) including the exact KiCad version, ViaStitching version, clearance value and, ideally, a minimal .kicad_pcb demonstrating the problem.

closed 2026-08-25 by weirdgyn