weirdgyn/viastitcher #30: KiCad 8 ignoring all other nets

burnt-past, 2024-03-19:
Running via stitching will keep vias inside copper pours but ignores all footprints and other nets.
[image]

snorkman88, 2024-08-23:
Any updates on this?

jmole, 2024-12-25:
I've updated the plugin to KiCAD 8 here: https://github.com/jmole/viastitching/tree/master

give it a try if you have some time and let me know if you still have the issue

weirdgyn, 2026-08-21:
Closing as resolved/superseded on current master. The overlap logic has since been substantially revised by PRs #37, #41, and #44 to handle overlapping zones, round/custom pads, tracks, vias, footprints, and other-net items. If the problem is reproducible with current master, please open a new issue with the board file and KiCad version.

closed 2026-08-21 by weirdgyn