weirdgyn/viastitcher #31: Downloaded package hash does not match the repository record on KiCad8

xinaovo, 2024-03-28:
When I tried to install this extension on KiCad 8.0.1, it failed to install and thrown this error message. It says the HASH of the downloaded package is not match the repository record. Besides, it warns me its not compatible with KiCad 8 when I tried to install it. 
Translated screenshot is shown below.
[image]
Original screenshot is shown below.
[image]

weirdgyn, 2024-03-28:
this's not my plugin... the functionality is similar but you're trying to
install a plugin from jsreynaud repository (that's not mine of course).

xinaovo, 2024-03-28:
> this's not my plugin... the functionality is similar but you're trying to install a plugin from jsreynaud repository (that's not mine of course). Il giorno gio 28 mar 2024 alle ore 13:10 xinaovo ***@***.***> ha scritto:
> […](#)
> -- bye by(t)e[s]... WeirdGyn!

Thanks for replying! Sorry for not checking the source of the extension... I'll try again with this repository later.

closed 2024-03-28 by xinaovo